alangarf/apple-one #17: build process does not work

OE1DEA, 2018-03-25:
dalb@latitude:~/apple-one/boards/tinyfpga_b2/yosys$ make
mkdir -p build
yosys -q -p "chparam -list; hierarchy -top apple1_top; synth_ice40 -blif build/apple1.blif" ../../../rtl/apple1.v ../../../rtl/clock.v ../../../rtl/pwr_reset.v ../../../rtl/ram.v ../../../rtl/rom_wozmon.v ../../../rtl/rom_basic.v ../../../rtl/cpu/arlet_6502.v ../../../rtl/cpu/arlet/ALU.v ../../../rtl/cpu/arlet/cpu.v ../../../rtl/uart/uart.v ../../../rtl/uart/async_tx_rx.v ../../../rtl/vga/vga.v ../../../rtl/vga/vram.v ../../../rtl/vga/font_rom.v ../../../rtl/ps2keyboard/debounce.v ../../../rtl/ps2keyboard/ps2keyboard.v ../../../rtl/boards/tinyfpga_b2/clock_pll.v ../../../rtl/boards/tinyfpga_b2/apple1_hx8k.v
/bin/sh: 1: yosys: not found
Makefile:20: recipe for target 'build/apple1.blif' failed
make: *** [build/apple1.blif] Error 127

Any idea what's going wrong?

al177, 2018-03-25:
It looks like yosys and possibly the rest of the icestorm tools aren't in your path.

What is the output of these commands:
```
which yosys icepack arachne-pnr
echo $PATH
```

And how did you install the toolchain?

OE1DEA, 2018-03-25:
Sorry, it was my fault. I didn't follow the full installation guide at http://www.clifford.at/icestorm.

closed 2018-03-25 by OE1DEA