alangarf/apple-one #20: Modify to load as a MiSTer core?

JackZielke, 2018-04-09:
Would anyone that has the knowhow be interested in converting this to be a core for [MiSTer](https://github.com/MiSTer-devel/Main_MiSTer/wiki)?  It runs on the DE10-Nano and provides HDMI output and USB keyboard input.  There is already an [Apple II+](https://github.com/MiSTer-devel/Apple-II_MiSTer) core.

alangarf, 2018-04-12:
PRs happily accepted. 😃 

Shouldn't be too difficult, but I haven't got the time myself.

alangarf, 2018-04-12:
Man, found on the Atari Forum someone has "merged" this core into their project and made a mess of the structure and dumped all the files into a random directory. :( Will make keeping that up to date hard.

JackZielke, 2018-04-20:
Was it [this one](https://github.com/Gehstock/Mist_FPGA/tree/master/Apple%20-%201_MiST)?  That is for MiST, which is pretty close to MiSTer.  I am hoping this will make it even easier to port to MiSTer (I do not have a MiST FPGA).

On a related note, I have been converting cassette save files back to the original BASIC code [here](https://linuxcoffee.com/apple1/software/).  I was envisioning using the On Screen Display to load text files as if they were being typed on the keyboard.

alanswx, 2019-08-14:
I started with the MiST version, so unfortunately the directories are all wrong:
https://github.com/MiSTer-devel/Apple-I_MiSTer

I added some code to load code from text files. It isn't fully working yet. I need to fix it.

alangarf, 2019-08-14:
Hi,

I'd appreciate it if the MiST project could get the Apple One code back inline with this repo. Dumping all the RTL into a single directory is kinda poor and doesn't allow any modifications to trickle back to this project.

While I understand the excitement in adding it to MiST my work appears to have been misrepresented.

Thanks,
-A.

alanswx, 2019-08-15:
Let me see if I can pull it back together to look more like yours. In general the code goes into https://github.com/miSTer-devel/ as it's own repo. There are very few (if any?) forks in the main repo. The naming convention of some of the directories are important for the scripts to auto upgrade cores from the RBF files.  Do you have a MiSTer yet? Are you interested in getting one?

alanswx, 2019-08-15:
Take another look. https://github.com/MiSTer-devel/Apple-I_MiSTer

I can pull in the rest of your files. We can merge my changes back into your head, and then I can check out a fork of yours maybe? I think I just need the "releases" folder in the top level so the scripts will find it.

alangarf, 2020-01-27:
Look good. Cheers.

closed 2020-01-27 by alangarf