alangarf/apple-one #22: Issue with PS/2 keyboard on Terasic DE0 Board

dennyalbert, 2018-04-12:
When I flash the sof-file everything works correctly. But when I flash the pof-file to keep the software after a power cycle the keyboard doesn't work correctly anymore. The buttons are mixed-up completely. Any idea what goes wrong here and how to fix? I use the Quartus II 13.1 (32-bit) Web Edition.

trcwm, 2018-04-12:
Hi @dennyalbert,

I haven't seen this issue before. My A1 boots correctly when I flash the configuration files.
Do you verify the flash contents after flashing? Could be your config flash is corrupt.
The other possibility it that Quartus does not produce the correct .pof file. 
Can you attach your .pof file?

/Niels.

dennyalbert, 2018-04-12:
Hi @trcwm,

attached please find my pof-file.
Yes I did a verify and everything seems to be ok.
 
[apple-one.zip](https://github.com/alangarf/apple-one/files/1904476/apple-one.zip)

Regards,

Denny

trcwm, 2018-04-13:
Your .pof file works perfectly on my DE0 system. I'm not sure how to help you further I'm afraid.

/Niels.

dennyalbert, 2018-04-13:
I have now tested the workflow under windows without success. Only to rule out an error with the operating system. Otherwise I use linux.

Could I have a problem with the hardware?
I have tested some examples, everything has worked without any problems.
If there is a problem with the memory, should not the verify lead to an error?

Is there a possibility to read the memory bit-by-bit and compare that with the pof-file?

dennyalbert, 2018-04-15:
Now I have read the configuration from the board and there are many differences between the two files. How can that happen, shouldn’t be the files identical?
Attached the two files.
[read&written.zip](https://github.com/alangarf/apple-one/files/1912964/read.written.zip)

trcwm, 2018-04-17:
Have you erased the flash chip before programming?

dennyalbert, 2018-04-29:
I had the opportunity to test on a second Terasic DE0 board and have the same issue.
I tried to establish a connection via the serial interface, unfortunately without success. Attached two screenshots. Do you have any idea why the connection does not work?
[image]
[image]

trcwm, 2018-04-29:
Are you using a RS232 compliant serial to USB cable? A normal TTL-level cable won't work.

trcwm, 2018-04-29:
Yes, the DE0 has a level translator (unlike many other boards) and it is directly compatible with an RS232 COM port.

I power cycle the DE0 when changing the position of the RUN/PROG switch. I have never seen it misbehave in either state. 

When I power up in the PROG state, the DE0 does nothing. When I power up in the RUN state (and the flash has been programmed), my A1 works as intended.

dennyalbert, 2018-04-29:
This is really annoying that two boards behave differently.
Just to clarify, do I only need to load the pof-file or do I have to do some other configurations like for unused pins?

trcwm, 2018-04-29:
To my knowledge there are no additional requirements to make the A1 run from config flash.
You board seems to work ok if the configuration was loaded directly from the PC. My best guess is that your config flash is not working properly.

dennyalbert, 2018-04-30:
This was also my first guess that the config flash is the problem. But the same issue with two new boards? Unfortunately the EPCS4 is no longer available.

dennyalbert, 2018-05-02:
Unfortunately I have the same problem with your files.

dennyalbert, 2018-05-03:
Now I have built a cable for the serial port as described in the picture. I can receive characters but not send. Is there something special to do? Can the PS/2 keyboard stay connected?
[image]
[image]

trcwm, 2018-05-07:
If I'm not mistaken, the code will only transmit the data sent to the VGA to the UART.
The functionality to get the keyboard data from the UART is disabled by default.
You have to set the 'ps2_select' value in apple1_de0_top.v to zero.

dennyalbert, 2018-05-18:
I'm getting closer to the problem. The keyboard (Fujitsu KB400) itself seems to be the problem.
If I supply the keyboard with voltage before the FPGA everything works.
The same effect occurs if I press and hold the reset button (Button 0) before switching on and release afterwards.
Any ideas how to fix the problem?

dennyalbert, 2018-05-20:
Now I was able to test with an old AT keyboard (PS/2 adapter) and can confirm my guess that the Fujitsu keyboard was the problem.

closed 2018-05-20 by dennyalbert

alangarf, 2018-05-20:
Glad you solved it! 👍 how annoying 😄