alangarf/apple-one #42: Merging Lawrie's work

goran-mahovlic, 2024-03-19:
Hi, could you please check this fork https://github.com/lawrie/apple-one 

Lawrie added some boards but unfortunately he is not longer with us, it would be shame to lose those changes.

alangarf, 2024-04-09:
Can do. Not sure what level they are working. But yeah I'll pull them in shortly,

alangarf, 2024-04-09:
100% sad to see he's left us.

alangarf, 2024-04-12:
I've made a PR, but Lawrie made changes to the base RTL files. It's been a while since I've played with this code and it looks like the RAM settings have been changed to allow for more RAM etc for the boards. Usually these changes are put in the board specific areas so I'll have to test what changed etc. But I have a PR now for the boards so they won't be lost.

https://github.com/alangarf/apple-one/pull/43

goran-mahovlic, 2024-04-12:
Thank you! Yes it want be lost and it will be more visible.